RobTillaart/INA219 #4: a few questions

immortal-sniper1, 2023-06-03:
I want to use multiple INA219 and i know all their addresses and they all have the same stings (shunt and voltage ranges).

INA219 INA(0x40);     // is the constructor IF i wanted to store them in an array should i make something like this?

INA219 vector_INA[10] = {  INA(0x40), INA(0x41), INA(0x42), INA(0x43)    };  // like this OR do i need to increment the INA name in the parentheses ? 

I want to access them like this:  vector_INA[ IC_number]  OR is there a better way to make a array from them ? 

BTW was this lib ever tested on a RP2040? asking since like the ESP32 it has multiple I2C busses and pins.

RobTillaart, 2023-06-03:
Thanks for the issue,
Will cone back to it asap

RobTillaart, 2023-06-03:
Not tested with RP2040 except compilation.

Array I need to verify, 
Is a good question to create an example for.
(Might take a few days)

immortal-sniper1, 2023-06-03:
ok thx
for the RP2040 it did compile on my end too but i did not test yet since you mentioned that for the ESP32 u need to set pins and i guess it is the same for the RP2040

RobTillaart, 2023-06-03:
Means you might need to add

```|| defined(ARDUINO_ARCH_RP2040)```

Where ESP32 is tested

RobTillaart, 2023-06-03:
Quick wrote a demo sketch, compiles, 
I have no hardware to test

```cpp
//
//    FILE: INA219_array.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo array of INA219 sensors
//     URL: https://github.com/RobTillaart/INA219

#include "INA219.h"
#include "Wire.h"

INA219 arr_ina[3] = { INA219(0x40), INA219(0x41), INA219(0x42) };

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("INA219_LIB_VERSION: ");
  Serial.println(INA219_LIB_VERSION);

  Wire.begin();
  //  Wire.begin( sda, scl );  //  

  for (int i = 0; i < 3; i++)
  {
    if (! arr_ina[i].begin())
    {
      Serial.print("Could not connect:  ");
      Serial.print(i);
      Serial.println(". Fix and Reboot");
    }
  }

  for (int i = 0; i < 3; i++)
  {
    arr_ina[i].setMaxCurrentShunt(5, 0.002);
    delay(1000);
    Serial.println(arr_ina[i].getBusVoltageRange());
  }
}

void loop()
{
  Serial.println("\n\t#\tBUS\t\tSHUNT\t\tCURRENT\t\tPOWER\t\tOVF\t\tCNVR");
  for (int i = 0; i < 3; i++)
  {
    Serial.print("\t");
    Serial.print(i);
    Serial.print("\t");
    Serial.print(arr_ina[i].getBusVoltage(), 2);
    Serial.print("\t\t");
    Serial.print(arr_ina[i].getShuntVoltage_mV(), 2);
    Serial.print("\t\t");
    Serial.print(arr_ina[i].getCurrent_mA(), 2);
    Serial.print("\t\t");
    Serial.print(arr_ina[i].getPower_mW(), 2);
    Serial.print("\t\t");
    Serial.print(arr_ina[i].getMathOverflowFlag());
    Serial.print("\t\t");
    Serial.print(arr_ina[i].getConversionFlag());
    Serial.println();
    delay(1000);
  }
  delay(1000);
}

//  -- END OF FILE --
```

RobTillaart, 2023-06-03:
**not** tested
The change in the .h file should be

```cpp
#if defined(ESP8266) || defined(ESP32) || (defined(ARDUINO_ARCH_RP2040) && !defined(__MBED__))
  bool      begin(uint8_t sda, uint8_t scl);
#endif
```

The change in the .cpp file should be

```cpp

#if defined(ESP8266) || defined(ESP32)

bool INA219::begin(const uint8_t sda, const uint8_t scl)
{
  _wire = &Wire;
  _wire->begin(sda, scl);
  return isConnected();
}

#elif defined(ARDUINO_ARCH_RP2040) && !defined(__MBED__)

bool INA219::begin(uint8_t sda, uint8_t scl)
{
  _wire->setSCL(scl);
  _wire->setSDA(sda);
  _wire->begin();
  return isConnected();
}
#endif
```

Note: the RP2040 compile environment (MBED or other) has different Wire interface/implementation.

RobTillaart, 2023-06-12:
@immortal-sniper1 
Any progress made?

immortal-sniper1, 2023-06-12:
YES tring to figure out if the board i made is shorted or not since atm I cant truly test the code since there is a power loop and i was stuck there since i cant have a input or output ( since the 2 are shorted ......)
And honestly i am not sure if a transistor is on in linear region or it is a proper short

RobTillaart, 2023-06-12:
OK, 
Hope you get the right insights soon, working with broken boards can consume a lot of time.

In the meantime I am going to make a develop branch + PR to improve the support for the RP2040 (as coded above).

immortal-sniper1, 2023-06-12:
ok THX.
And yes i lost a lot of time trying to figure out what is wrong (besides other stuff that apparently uses way more power then i thought but still acceptable)  
well such is the life of a V1 board

RobTillaart, 2023-06-13:
Released the 0.1.4 version with RP2040 code (and some minor edits) from post above.
You should now be able to set the I2C pins.

RobTillaart, 2023-06-28:
@immortal-sniper1 
If there are no questions left, please close the issue

closed 2023-06-28 by immortal-sniper1