RobTillaart/INA219 #6: Question about speed

N0ury, 2023-09-23:
Hi Rob this is just a question.
What would be the highest frequency on an ESP32-S2 with PSRAM for a loop with:
- getCurrentLSB()
- store result in PSRAM
Nothing more.
Can it grow up to 1MHz?
My need is to detect very small peaks in power consumption measurement.

Eventually same question with an ESP32-S3 but with a Serial.println (no PSRAM), or a picow (RP2040).

RobTillaart, 2023-09-23:
Hard to say as there are many factors

A quick and dirty calculation:

clock speed the INA219 can handle 2.5 MHz in theory (never tested)
As the INA219 can do 2.5 MHz that implies max 2.500.000 / 8 bytes = 300.000 bytes per second
as the current is a float of 4 bytes there are max 300.000 / 4 = 75000 floats per second.

We did not count any overhead or further processing so lets say that is a factor 3
that would leave 25000 measurements per second max. 

That is a factor 40 from your 1 MHz samples.

Think the INA219 is not the sensor that can do this.

-----

> My need is to detect very small peaks in power consumption measurement.

Questions:
- what are the durations of such peaks?
- what is the normal load? AC/DC?
- what is the peak load?
- how often do you have the peaks? Poison distribution, random, uniform?

These questions should better be discussed on the Arduino forum or so. 
I think you must create a hardware solution that triggers and hold when a peak occurs. (interrupt).
And that can be reset from the MCU
Guess you should look for a comparator (convert the current peak to a voltage peak) 
+ some interrupt handling

RobTillaart, 2023-09-23:
You should get a book like - https://www.amazon.com/Practical-Electronics-Inventors-Fourth-Scherz/dp/1259587541 
I learned a lot from it and although some parts might be difficult at first you will learn a lot too

N0ury, 2023-09-23:
Thanks for your advice.
I'll think about it.
I've tried an INA138, an INA226, and a MAX4372.
None of them gave me satisfaction. Either there was too much noise, or the number of samples per second was low.
I'm going to start by putting things straight, and in the meanwhile start reading this book (thanks for the tip).

closed 2023-09-23 by N0ury

RobTillaart, 2023-09-23:
https://www.youtube.com/watch?v=ic_yEUV7Y3c