RobTillaart/INA219 #11: Confusion on setting the mask using setBusADC()/setShuntADC()

Darkextratoasty, 2024-01-16:
I'm a bit confused on how to set the bits for the resolution and averaging using the `setBusADC()` and `setShuntADC()` functions. From reading the datasheet, it seems to me that I need to first set the resolution using the 0x00 through 0x03 options, then I need to set the averaging samples using the 0x08 through 0x0F options. So to set the shunt ADC resolution to 12 bits and 128 samples, I would use:
```
INA.setShuntADC(0x03); //12 bits
INA.setShuntADC(0x0F); //128 samples
```
However, if I then measure the amount of time it takes to perform a current reading, which should take about 70mS, it only takes ~85uS to take the measurement, meaning it's actually on 9 bits and 1 sample. I've also tried just using 0x0F, but that has the same result. No matter what I feed into the `setShuntADC()` function (as long as it's <=0x0F) the time it takes to do a measurement doesn't change. Here's the example code that shows this:
```
#include "INA219.h"
#include "Wire.h"

INA219 INA(0x45); // I've changed the address on my module

void setup(void) {
  Serial.begin(115200);
  while (!Serial)
    delay(1); //wait for serial monitor

  Serial.print("\n\n\n\n\n\nINA219_LIB_VERSION: ");
  Serial.println(INA219_LIB_VERSION);

  Wire.begin();
  Wire.setClock(800000);
  if (!INA.begin() ) {
    Serial.println("could not connect. Fix and Reboot");
  }

  INA.setMaxCurrentShunt(3.2, 0.1);
  INA.setGain(8);

  INA.setShuntADC(0x03); //set resolution to 12 bits
  INA.setShuntADC(0x0F); //set samples to 128
}

void loop(void)  {
  float current = 0;

  unsigned long start = micros();
  current = INA.getCurrent();
  unsigned long end = micros();

  Serial.print("Time: \t"); Serial.print(end-start); Serial.print("uS");
  Serial.println();
  Serial.print("Current:  \t"); Serial.print(current, 4); Serial.print("A \t");
  Serial.println();

  delay(2000);
}
```
Am I completely misunderstanding how to use these functions?

RobTillaart, 2024-01-17:
Thanks for the questions,
it will take several days before I have time to investigate, need to check if I still have such sensor somewhere.

RobTillaart, 2024-01-17:
A quick look showed a TODO note in the .cpp so I label it as bug to give it some priority.

RobTillaart, 2024-01-17:
had a brief moment in the datasheet.

```cpp
  INA.setShuntADC(0x03); //set resolution to 12 bits
  INA.setShuntADC(0x0F); //set samples to 128
```

Only the second call is needed, when setting the samples the resolution is always 12 bit (given the duration).

I will implement 4 additional functions (and add some documentation)
```cpp
bool setBusResolution(uint8_t bits);    //  9..12 otherwise false

bool setBusSamples(uint8_t value);      //  value = 0..7 => 2^value samples

bool setShuntResolution(uint8_t bits);  //  9..12 otherwise false

bool setShuntSamples(uint8_t value);    //  value = 0..7 => 2^value samples 
```

Expect a develop branch later today
I do not have time to setup hardware, so if you have time to test it would be appreciated.

RobTillaart, 2024-01-17:
@Darkextratoasty 

Looking at you code, I saw 
```cpp
 Wire.setClock(800000);
```
Can you change it to 100000?

Datasheet states it should be supported (see below) but these are often only under ideal circumstances.

_The INA219 supports the transmission protocol for fast (1- to 400-kHz) and high-speed (1-kHz to 2.56-MHz) modes. All data bytes are transmitted most significant byte first._

RobTillaart, 2024-01-17:
Created the develop branch with the four new functions - https://github.com/RobTillaart/INA219/tree/develop
It includes the checking of the **writeRegister()** calls.
PR will be made asap.

Going through the code this morning did not reveal any code bug (absence of proof != proof of absence)

Darkextratoasty, 2024-01-17:
First off, thank you for being so quick to respond and helpful.

I tested a few different INA219 breakout board, a few different I2C cables, and a few different microcontrollers just to be sure it isn't a hardware issue.

First changed the clock speed to 100KHz while still on the master branch. The 85uS turns into about 520uS and changing the value fed into `setShuntADC()` still has no effect on the time. `getShuntADC()` seems to be indicating that the register is writing correctly however. 

After moving to the develop branch, I used
```
INA.setShuntResolution(9);
Serial.println(INA.getShuntADC(), HEX);
```
which printed out `0` as expected and took about 520uS for an `INA.getCurrent()` action. I then changed the resolution to 12
```
INA.setShuntResolution(12);
Serial.println(INA.getShuntADC(), HEX);
```
which printed `3` and took 520uS to get current. Setting the samples to 1
```
  INA.setShuntResolution(12);
  INA.setShuntSamples(1);
  Serial.println(INA.getShuntADC(), HEX);
```
printed `9` and took 520uS. Setting samples to 7 (128 samples)
```
  INA.setShuntResolution(12);
  INA.setShuntSamples(7);
  Serial.println(INA.getShuntADC(), HEX);
```
prints `F` and still takes about 520uS.

So the `getShuntADC()` is indicating that the register is being correctly written, but the time it takes to perform a reading doesn't match up. I'm wondering if perhaps the actual time it's taking to make a measurement is changing, but the time that I'm seeing with `getCurrent()` is only the time it takes to read the latest measurement from a buffer on the INA219. I tried looping as fast as possible 100 times, reading the current with `getCurrent()` each time, which ran through in about 52mS, indicating each reading still took around 520uS. However, it's possible the actual value is only updating every ~62mS (for 12bit 128sample), and I'm just reading the same value over and over. I plan to test this out, but first I need to come up with a load that changes current fast enough for it to change within 520uS. I'll update when I get that rigged up, hopefully today.

RobTillaart, 2024-01-17:
@Darkextratoasty 

Interesting line in the datasheet. It mentions the resolution and samples are for the shunt VOLTAGE register.
(idem for BUS voltage register)

[image]

You are timing **getCurrent()**, so can you try to time **getShuntVoltage()**?

To be continued..

Darkextratoasty, 2024-01-17:
It appears that my guess was correct, the INA219 will let you read as fast as the I2C bus can go, but the values will only update at the rate determined by the resolution and number of samples. I made a simple quickly changing load using a resistive load connected to a mosfet that's driven by another arduino running a sped up "fade" example sketch. This is the sketch I used to read the INA219 to test this (on an esp32 board this time):
```
#include "INA219.h"
#include "Wire.h"

INA219 INA(0x40);

void setup(void) {
  Serial.begin(115200);

  delay(7000);
  Serial.print("\n\n\n\n\n\nINA219_LIB_VERSION: ");
  Serial.println(INA219_LIB_VERSION);

  Wire.begin();
  Wire.setClock(100000);
  if (!INA.begin() ) {
    Serial.println("could not connect. Fix and Reboot");
  }

  INA.setMaxCurrentShunt(3.2, 0.1);
  INA.setGain(8);

  INA.setShuntResolution(9);
//  INA.setShuntSamples(7);
  Serial.print("getShuntADC result: 0x0");
  Serial.println(INA.getShuntADC(), HEX);

  float current[512] = {0};
  unsigned long start = micros();
 
  for(int i = 0; i < 512; i++) {
    current[i] = INA.getCurrent_mA();
  }
  unsigned long end = micros();

  Serial.print("Time total: \t"); Serial.print((end-start)/1000); Serial.print("mS  \tTime per reading: \t"); Serial.print((end-start)/512); Serial.println("uS");
  Serial.println();
  Serial.println("Current:  \t");// Serial.print(current, 4); Serial.print("A \t");
  for(int i = 0; i < 32; i++) {
    for(int j = 0; j < 16; j++) {
      Serial.print(current[(i*16) + j]);
      Serial.print(" \t");
    }
    Serial.println();
  }
  Serial.println();
}

void loop(void)  {}
```
With resolution set to `9` and samples unset, the sample time should be 84uS, while the I2C read time is about 600uS (esp32 is apparently slower than RP2040), so I would expect to see a different value for each reading. The output shows just that:
```
INA219_LIB_VERSION: 0.2.1
getShuntADC result: 0x00
Time total: 	307mS  	Time per reading: 	599uS

Current:  	
263.87 	263.87 	263.87 	395.51 	502.93 	503.32 	502.93 	106.15 	503.32 	502.15 	502.54 	503.91 	9.86 	503.91 	502.54 	503.32 	
502.54 	303.42 	503.91 	503.32 	503.32 	502.93 	504.30 	503.32 	502.54 	503.32 	502.54 	502.54 	502.93 	503.32 	503.32 	502.93 	
502.54 	503.32 	503.32 	502.93 	499.71 	503.32 	503.32 	502.54 	502.54 	503.32 	502.93 	-0.39 	501.76 	-1.17 	107.52 	0.00 	
-0.39 	0.00 	0.00 	0.39 	-0.39 	-0.39 	-0.78 	-0.78 	-0.39 	0.00 	-0.39 	0.00 	0.39 	0.39 	0.39 	0.00 	
-0.78 	0.88 	-0.39 	0.00 	0.39 	0.39 	214.45 	-0.39 	0.00 	-0.39 	-0.78 	530.96 	-0.39 	0.00 	-0.39 	0.00 	
426.07 	-0.39 	0.39 	0.39 	0.39 	0.00 	201.27 	-0.39 	-0.39 	0.88 	0.39 	-7.81 	0.39 	0.00 	51.07 	-1.17 	
0.00 	0.00 	-0.78 	314.06 	-0.39 	0.00 	-0.78 	-0.78 	505.08 	0.39 	0.00 	0.39 	0.00 	503.32 	0.00 	-0.39 	
-0.39 	0.00 	-0.39 	502.15 	-0.39 	-0.39 	0.39 	-0.78 	422.36 	-0.39 	-0.39 	244.92 	-0.39 	212.40 	0.00 	0.39 	
475.00 	0.00 	-8.59 	-0.39 	0.39 	504.30 	-0.39 	0.88 	0.39 	0.00 	502.54 	0.00 	0.00 	-0.39 	-0.39 	0.00 	
501.37 	0.88 	0.39 	-0.39 	0.00 	500.98 	-0.39 	0.39 	429.39 	0.39 	500.59 	-0.78 	0.39 	523.54 	-0.39 	412.50 	
-0.39 	0.00 	504.30 	0.00 	206.64 	0.39 	-0.39 	502.54 	502.54 	0.00 	1.27 	-0.39 	0.39 	502.15 	0.39 	-0.39 	
-1.56 	0.39 	501.37 	-1.17 	-0.39 	337.99 	0.39 	500.98 	-0.39 	-0.39 	505.47 	0.00 	501.37 	0.88 	0.39 	503.32 	
[... 77 lines cut ...]
341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	
341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	
341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	
341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	
341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	
341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	
341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	341.70 	361.62 	
361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	
361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	
361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	
361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	
361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	
361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	
361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	361.62 	96.97 	96.97 	
96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	
96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	
[... cut ...]
96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	
96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	96.97 	

```
All this to say, the register set is working correctly (I suspect the master branch library was also working correctly, I'll test that out soon), but the INA219 will let you read old values as fast as the I2C bus can go. 

Per your most recent comment, `getCurrent()` and `getShuntVoltage()` yield the same results with respect to the timing.

Darkextratoasty, 2024-01-17:
Yep, the master branch library was fine, `setShuntADC()` works as intended.

RobTillaart, 2024-01-17:
That is an amazing analysis, well done.
In short one configures the internal read process, the values in the registers are "just" the last valid values.
Tomorrow I will think how to describe this in the readme.md file,
Again thanks for your tests and work, really appreciated 👍

RobTillaart, 2024-01-18:
This is for now the updated section that will be in the readme.md.

#### Configuration BUS and SHUNT

**Note:**
The internal conversions runs in the background in the INA219.
If a conversion is finished the measured value is stored in the appropriate register. 
The last obtained values can always be read from the registers, so they will not block.
Result can be that you get the very same value if no new data is available yet.
This is especially true if you increase the number of samples.
(See also discussion in #11).

Using more samples reduces the noise level, but one will miss the faster 
changes in voltage or current.
Depending on your project needs you can choose one over the other.

As a rule of thumb on could take the time between two I2C communication time 
for reading a register as an upper limit.
This would give a fresh reading every time. 
NB it is always possible to average readings fetched from the device.

Use one of these three so set **bus** resolution and sampling.

- **bool setBusResolution(uint8_t bits)** bits = 9..12, always 1 sample.
Returns false if parameter out of range.
Returns false if it could not write settings to device.
- **bool setBusSamples(uint8_t value)** value = 0..7 => maps to 2^value samples.
Always 12 bits.
Returns false if parameter out of range.
Returns false if it could not write settings to device.
- **bool setBusADC(uint8_t mask = 0x03)** see table below. 
Check datasheet for all details.
Returns false if parameter out of range (mask > 0x0F).
Returns false if it could not write settings to device.
- **uint8_t getBusADC()** returns mask, see table below.

Use one of these three so set **shunt** resolution and sampling.

- **bool setShuntResolution(uint8_t bits)** bits = 9..12, always 1 sample.
Returns false if parameter out of range.
Returns false if it could not write settings to device.
- **bool setShuntSamples(uint8_t value)** value = 0..7 => maps to 2^value samples.
Always 12 bits.
Returns false if parameter out of range.
Returns false if it could not write settings to device.
- **bool setShuntADC(uint8_t mask = 0x03)** see table below. 
Check datasheet for all details.
Returns false if parameter out of range (mask > 0x0F).
Returns false if it could not write settings to device.
- **uint8_t getShuntADC()** returns mask, see table below.

#### Resolution samples table

mask = both resolution + averaging multiple samples.
minus - == don't care

|  bit mask  |  value  |  resolution  |  samples      | conversion time |
|:----------:|:-------:|:-------------|:--------------|:---------------:|
|    0-00    |  0 / 4  |     9 bit    |    1 sample   |      84 μs      |
|    0-01    |  1 / 5  |    10 bit    |    1 sample   |     148 μs      |
|    0-10    |  2 / 6  |    11 bit    |    1 sample   |     276 μs      |
|    0-11    |  3 / 7  |    12 bit    |    1 sample   |     532 μs      |
|            |         |              |               |                 |
|    1000    |    8    |    12 bit    |    1 sample   |     532 μs      |
|    1001    |    9    |    12 bit    |    2 samples  |    1.06 ms      |
|    1010    |   10    |    12 bit    |    4 samples  |    2.13 ms      |
|    1011    |   11    |    12 bit    |    8 samples  |    4.26 ms      |
|    1100    |   12    |    12 bit    |   16 samples  |    8.51 ms      |
|    1101    |   13    |    12 bit    |   32 samples  |   17.02 ms      |
|    1110    |   14    |    12 bit    |   64 samples  |   34.05 ms      |
|    1111    |   15    |    12 bit    |  128 samples  |   68.10 ms      |

- note that a new value set can substantially increase the conversion time.
- note that you cannot set e.g. 9 bits and 16 samples.
- note that there are 3 ways to set 12 bits 1 sample.

Darkextratoasty, 2024-01-18:
That is a very clear description, thanks you for your work developing and maintaining this library

RobTillaart, 2024-01-18:
Merged and released 0.2.1 version, so I close this issue.

closed 2024-01-18 by RobTillaart