RobTillaart/INA219 #13: wrong reading

immortal-sniper1, 2024-02-05:
So here is my setup:
Arduino uno r4 
INA219  INA_ADDRESS[5] = {0x00, 0x40, 0x41, 0x44, 0x45 };   //// all work well , tested with RPI4 +python , so HW is ok 

my code:
#include <Wire.h>
#include <PCA95x5.h>
#include "INA219.h"
#include <DAC121.h>

INA219 INA1(0x40);                                                           // 1-4   0x40 0x41 0x44 0x45    (0x46 for the SDA bodged one! cell_3)
INA219 INA2(0x41); 
INA219 INA3(0x44); 
INA219 INA4(0x45); 

static const uint8_t INA_ADDRESS[5] = {0x00, 0x40, 0x41, 0x44, 0x45 };      // 1-4   0x40 0x41 0x44 0x45

/*
ICs used with I2C:
INA219                        https://github.com/RobTillaart/INA219
PCA9535                       https://github.com/semcneil/PCA95x5
DAC121C085CIMM/NOPB           https://github.com/Wh1teRabbitHU/DAC121C08x-Driver
TCA9546A (mux board)          https://github.com/closedcube/ClosedCube_TCA9546A_Arduino
PIDcontroller lib             https://github.com/DonnyCraft1/PIDArduino
TPS63020DSJR SMPS             https://www.ti.com/lit/ds/symlink/tps63021.pdf?ts=1705144802509&ref_url=https%253A%252F%252Fwww.google.com%252F
                              https://www.ti.com/lit/ug/slva726/slva726.pdf?ts=1705152730658

*/

//----------------------------------------------------------------------------------------------------------------------------------------------
//---------------------------------------------------------- MAIN LOOPS ------------------------------------------------------------------------
//----------------------------------------------------------------------------------------------------------------------------------------------

void setup()
{
  Serial.begin(9600); // open the serial port at 9600 bps:

  Wire.begin(); //SDA & SDL

}

void loop()
{

  Serial.println(" Cell NO 1  ");
  Last_Cell_Voltage[1] = INA1.getBusVoltage();
  Last_Cell_Current[1] = INA1.getCurrent(),
  Serial.print(  Last_Cell_Voltage[1], 4);
  Serial.print(" V   ");
  Serial.print(  Last_Cell_Current[1], 4);  // should be 0 atm
  Serial.println(" A   ");
  Serial.println("   ");
  Serial.println("   ");
  Serial.println(INA1.getBusVoltage_mV(),3 );
  Serial.println("   ");
  Serial.println("   ");
  
  

  Serial.println(" Cell NO 2  ");
  Last_Cell_Voltage[2] = INA2.getBusVoltage();
  Last_Cell_Current[2] = INA2.getCurrent(),
  Serial.print(  Last_Cell_Voltage[2], 4);
  Serial.print(" V   ");
  Serial.print(  Last_Cell_Current[2], 4);  // should be 0 atm
  Serial.println(" A   ");
  Serial.println("   ");
  Serial.println("   ");

    Serial.println(" Cell NO 3  ");
  Last_Cell_Voltage[3] = INA3.getBusVoltage_mV();
  Last_Cell_Current[3] = INA3.getCurrent_mA(),
  Serial.print(  Last_Cell_Voltage[3], 4);
  Serial.print(" mV   ");
  Serial.print(  Last_Cell_Current[3], 4);  // should be 0 atm
  Serial.println(" mA   ");
  Serial.println("   ");
  Serial.println("   ");

  Serial.println(" Cell NO 4  ");
  Last_Cell_Voltage[4] = INA4.getBusVoltage_mV();
  Last_Cell_Current[4] = INA4.getCurrent_mA(),
  Serial.print(  Last_Cell_Voltage[4], 4);
  Serial.print(" mV   ");
  Serial.print(  Last_Cell_Current[4], 4);  // should be 0 atm
  Serial.println(" mA   ");
  Serial.println("   ");
  Serial.println("   ");

  delay(30000);

}

all i get in the terminal is o all over the place 
  Serial.println(INA1.getBusVoltage_mV(),3 );  this outputs -100000.0   for some reason 

can you please help me? i am not sure if it from the library or my code

immortal-sniper1, 2024-02-05:
if you need more info please ask and ill post it here

RobTillaart, 2024-02-05:
Thank you for reporting this question.
As I'm quite busy it might take several days before I can investigate.

Quick look => you need to add some initialization, like

```cpp
  if (INA1.begin() == false) Serial.println("Could not connect:  1. Fix and Reboot");
  if (INA2.begin() == false) Serial.println("Could not connect:  2. Fix and Reboot");
  if (INA3.begin() == false) Serial.println("Could not connect:  3. Fix and Reboot");
  if (INA3.begin() == false) Serial.println("Could not connect:  4. Fix and Reboot");

   INA1.setMaxCurrentShunt(5, 0.010);
   INA2.setMaxCurrentShunt(5, 0.010);
   INA3.setMaxCurrentShunt(5, 0.010);
   INA4.setMaxCurrentShunt(5, 0.010);
```

immortal-sniper1, 2024-02-05:
THX 
i will try that fix and report here,
i can wait a few days if you are that busy

RobTillaart, 2024-02-06:
@immortal-sniper1 
Can you try the basic single device example?
Please post its output,

As I do not have an UNO R4 I cannot check if the board is playing up in some way,

Do you have pull up resistors on the SDA and SCL lines?
Which value?

RobTillaart, 2024-02-06:
What is the length of the wires between board and sensors?

immortal-sniper1, 2024-02-06:
[image]
  this is the output from after i added the initialisation you told me yesterday

immortal-sniper1, 2024-02-06:
> What is the length of the wires between board and sensors?

10*cm more or less as for pull ups i am useing the internal ones , another I2C IC seems to work fine in the same cconditions

immortal-sniper1, 2024-02-06:
Bus Voltage: 3.820 V
Supply Voltage: 3.820 V
Bus Current: 0.598 ma
Power: 2.439 mW

Bus Voltage: 0.856 V
Supply Voltage: 0.856 V
Bus Current: 0.000 ma
Power: 0.000 mW

Bus Voltage: 0.860 V
Supply Voltage: 0.860 V
Bus Current: 0.098 ma
Power: 0.000 mW

Bus Voltage: 0.844 V
Supply Voltage: 0.844 V
Bus Current: 0.000 ma
Power: 0.000 mW

this is the expected output read with python  NOTE: the .844V is through 2Mohm so it may change depending on sampling rate 
cell 1 at 0x40 is the only one reading real data atm and 3.8V is what the multi meter confirmed to be well

immortal-sniper1, 2024-02-06:
[image]
demo code output

RobTillaart, 2024-02-06:
>  demo code output

States : **Could not connect. Fix and reboot**  so that data has no meaning. 

float INA219::getBusVoltage()
{
  uint16_t value = _readRegister(INA219_BUS_VOLTAGE);
  uint8_t flags = value & 0x03;
  //  math overflow handling
  if (flags & 0x01) return -100;

(Better copy the text instead of posting images, easier to refer to)

----

> this is the output from after i added the initialisation you told me yesterday

immortal-sniper1, 2024-02-06:
ok i will then edit the lib .cpp file and then run the code and paste it here

RobTillaart, 2024-02-06:
> ok i will then edit the lib .cpp file and then run the code and paste it here

The sensor returns an error, apparently it cannot measure the busvoltage correctly.
What change do you have in mind?

immortal-sniper1, 2024-02-06:
reading datasheet now

immortal-sniper1, 2024-02-06:
``` If larger full-scale shunt voltages are desired, the INA219 provides a PGA function that increases the full-scale
range up to 2, 4, or 8 times (320 mV). Additionally, the bus voltage measurement has two full-scale ranges: 16 or
32 V````  this is not set right this is my guess

RobTillaart, 2024-02-06:
use  the getters to know the actual settings,

(from ina219.h)
```cpp
  //  voltage = 16, 32  (values below 32 are rounded to 16 or 32)
  bool     setBusVoltageRange(uint8_t voltage = 16);
  uint8_t  getBusVoltageRange();  //  returns 16 or 32.

  //  factor = 1, 2, 4, 8
  bool     setGain(uint8_t factor = 1);
  uint8_t  getGain();
```

RobTillaart, 2024-02-06:
The python code does not set any gain so it uses the defaults too...

immortal-sniper1, 2024-02-06:
maybe the lib set it to a better default , you also set in somewhere at something i did not reverse  engineer out yet

RobTillaart, 2024-02-06:
> Could not connect. Fix and Reboot
32

The sensor is not seen.

Can you check with an I2C scanner if the device is visible at all?

immortal-sniper1, 2024-02-06:
i can try , there is another I2C IC i can test to see if it is  a buss problem or a chip problem

immortal-sniper1, 2024-02-07:
it worked a bit it may be improper contact somewhere 
UPDATE: so it is a wire contact problem , after that is fixed your demo code seems to run well 

my test code not so much will try to add a 1s delay after setting the max current like you did

RobTillaart, 2024-02-07:
Good to hear it starts working, the initialization is mandatory to get meaningful numbers.

immortal-sniper1, 2024-02-07:
it was not in the examples so they is why i did not add it , also no idea what is the default value

RobTillaart, 2024-02-07:
Then I need to check examples (maybe I mix up with other library, have seen tons of code last two months)

RobTillaart, 2024-02-07:
There are three examples with an initialization and two examples without.

If you only need the voltage you do not need the init
If you want the current you need to set the 
**setMaxCurrentShunt(5, 0.010)**

immortal-sniper1, 2024-02-08:
i also tried with a delay after setting the shunt and max current but still it did not work

RobTillaart, 2024-02-08:
> i also tried with a delay after setting the shunt and max current but still it did not work

First test is to get connection

```cpp
//
//    FILE: INA219_connected.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo array of INA219 sensors
//     URL: https://github.com/RobTillaart/INA219

#include "INA219.h"

INA219 arr_ina[16] =
{
  INA219(0x40), INA219(0x41), INA219(0x42), INA219(0x43),
  INA219(0x44), INA219(0x45), INA219(0x46), INA219(0x47),
  INA219(0x48), INA219(0x49), INA219(0x4A), INA219(0x4B),
  INA219(0x4C), INA219(0x4D), INA219(0x4E), INA219(0x4F),
};

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("INA219_LIB_VERSION: ");
  Serial.println(INA219_LIB_VERSION);

  Wire.begin();

  for (int i = 0; i < 16; i++)
  {
    if (! arr_ina[i].begin())
    {
      Serial.print("Could not connect:  ");
      Serial.print(i);
      Serial.print("\t");
      Serial.print(arr_ina[i].getAddress(), HEX);
      Serial.println();
    }
    else
    {
      Serial.print("Found: ");
      Serial.print(arr_ina[i].getAddress(), HEX);
      Serial.println();
    }
  }
}

void loop()
{
}

//  -- END OF FILE --
```

If no device is found then there is a hardware problem.

immortal-sniper1, 2024-02-09:
output from the above code:

INA219_LIB_VERSION: 0.2.1
Found: 40
Could not connect:  1    41
Could not connect:  2    42
Could not connect:  3    43
Could not connect:  4    44
Could not connect:  5    45
Could not connect:  6    46
Could not connect:  7    47
Could not connect:  8    48
Could not connect:  9    49
Could not connect:  10    4A
Could not connect:  11    4B
Could not connect:  12    4C
Could not connect:  13    4D
Could not connect:  14    4E
Could not connect:  15    4F

RobTillaart, 2024-02-09:
@immortal-sniper1 
Please post the output

immortal-sniper1, 2024-02-10:
INA219_LIB_VERSION: 0.2.1
Found: 40
Could not connect:  1    41
Could not connect:  2    42
Could not connect:  3    43
Could not connect:  4    44
Could not connect:  5    45
Could not connect:  6    46
Could not connect:  7    47
Could not connect:  8    48
Could not connect:  9    49
Could not connect:  10    4A
Could not connect:  11    4B
Could not connect:  12    4C
Could not connect:  13    4D
Could not connect:  14    4E
Could not connect:  15    4F

immortal-sniper1, 2024-02-10:
also i noticed this:
If i flash the board and reset it does not work 
but if i flash then reset then disconnect GND between boards , reconnect it and then reset it works

RobTillaart, 2024-02-10:
That does not sound like a library problem...

immortal-sniper1, 2024-02-10:
yea ,
the signals look ok on a oscilloscope but maybe it is an intermittent contact problem or something like that 
so for now until i am sure of it ill maybe close this issue (since i am sort of convinced it in on the HW connectivity side 
do u agree?

RobTillaart, 2024-02-10:
Don't know for sure if it is a bad contact, I've had a failing power supply once that caused intermittent issues. 
Since then I don't trust any component (including my libs) for 100% :)

When something is not working I always go to the simplest setup and build up complexity from there.
If the most simple version does not work, that often is caused by broken hardware, bad contacts, no pull ups where needed, swapped pins, actual use of other device (trying to talk I2C to a SPI device), power supplies, interference from outside etc.

Still I am interested in the output of the latest sketch I posted. (that should show voltage for found devices)

And yes, you can always close the issue and reopen when needed (it is your issue).

RobTillaart, 2024-02-10:
> the signals look ok on a oscilloscope

At which end of the wire?

immortal-sniper1, 2024-02-10:
> > the signals look ok on a oscilloscope
> 
> At which end of the wire?

it was on the receiveing board connector there were a lot of artefacts but i am not sure if it was bad scope or bad signaling but the signals themselves looked ok maybe it is just random noise i pick with wires and a breadboard

RobTillaart, 2024-02-14:
Any progress or insights?

immortal-sniper1, 2024-02-14:
well ina code is working periodically 
another IC lib was   shifted by 2 bits from 1 point onwards and that caused problems on 1/2 the board and then there was a problem with my voltage reference 
So ok...  figured out problems and solved some of them 
and it all works fine with a RPI so i blame some /most of the problem on the HW side , mostly the connectivity atm 
regarding the library for now i think it is all ok ( until i restart testing with some extreme connectivity solution like soldering wires to the terminals and if that does not fix it i dont know what is the problem ......

RobTillaart, 2024-02-14:
So if the library is not the problem, you may close the issue. (And reopen if needed).

closed 2024-02-15 by immortal-sniper1