RobTillaart/INA219 #14: Unsure if the calibration calculation used in the code is correct

ahmed-tkhan, 2024-03-15:
`  calib = round(0.04096 / (_current_LSB * shunt));
`
[image]

You use **round** in your calculation for the calibration but the datasheet says to use **truncate**. I don't know if this is an oversight or intentional. 

Thank you for the nice library. 👍

RobTillaart, 2024-03-15:
Thank you for pointing to this bug. 
This would be a systematic error (in 50% of the cases) so needs to be fixed.
Have you an indication about the impact, in absolute sense?  

> I'm nitpicking I know.

I would call it a decent review, I assume I missed it when implementing and it just stayed in the library since then.

RobTillaart, 2024-03-15:
Created a develop branch for version 0.3.0
includes the fix + some edits and build-CI update that were pending.

RobTillaart, 2024-03-15:
There is a problem with the pull request build-CI, one version just builds correctly, the other blocks/freezes.
DO not know the cause, however given that one build is OK you could try the develop branch if you want to.

closed 2024-03-16 by RobTillaart

RobTillaart, 2024-03-16:
Merged into master and released as 0.3.0

commit b36b5b34 (2024-03-16): Fix #14, round => truncate (#15)

- Fix #14, change round to truncate.
- update GitHub actions to v4
- update json check to v2
- update readme.md
- minor edits