RobTillaart/INA219 #16: Issue with getPower function

ChrisRed255, 2024-04-22:
Hi there,

I am reporting a bug that can mislead calculations while using INA219 sensor.

I had an overflow situation when the int16_t value is multiplied by 20 with the returned result : return (value * 20) * _current_LSB, in my example here is the issue : 
- Value is an int16 type.
- Value = 142B (hex value) = 5163 (decimal).
- Value * 20 = 103260, which is bigger than the limit of an int16 type which is 65535.
- In this case, the result is falsified by this overflow situation.

Suggested solution:
To solve this issue, we need to invert the multiplication, like this for example : return value * _current_LSB * 20. This way even if the register value (value) is big, when we multiply it by _current_LSB which is so small value that it doesn't affect the result because _current_LSB = maxcurrent / 2^15 = maxcurrent / 32768 = 0.00.... and if we multiply this result by 20, it's still not affected by an overflow.

RobTillaart, 2024-04-22:
I checked and there is a similar problem in INA226

```cpp
float INA226::getPower()
{
  uint16_t val = _readRegister(INA226_POWER);
  return val * 25 * _current_LSB;  //  fixed 25 Watt
}
```

ChrisRed255, 2024-04-22:
Yeah, you are right.

RobTillaart, 2024-04-22:
Issue solved with PR #17 

Again thanks for reporting!

closed 2024-04-22 by RobTillaart