RobTillaart/INA219 #19: Shutvoltage() is always returning positive values

aguilerabr, 2024-08-07:
Variable definition inside getShuntVoltage() should be type int16_t instead uint16_t for value.

RobTillaart, 2024-08-07:
Thanks for this issue, 

As I am busy it will take several days maybe weeks to dive into the datasheet and code. Labelled the issue as bug as it could be one and give it priority it needs.

Saw you created a PR which is appreciated! Will enable the build asap.

RobTillaart, 2024-08-07:
@aguilerabr 
Had a quick look in the datasheet and you are right, the register is signed so your fix makes sense.

Please note there are some open ends with the shunt voltage.
Especially table 7 indicates that depending on the GAIN set the max shunt voltage differs. (out of range gray area).
A function **bool shuntVoltageOverflow()** that returns true if last measurement is in gray area. This needs a private bool flag to indicate (possible) overflow when reading the shunt voltage register. To determine this overflow one needs to have the gain factor (or a derived value).

A function **int getMaxShuntVoltage()** returning 40 80 180 or 320 depending on gain, could be useful in this context.

closed 2024-08-14 by RobTillaart