RobTillaart/INA219 #22: Suitable for 4-20mA?

Mark-Wills, 2024-12-27:
Hello!

This is a question, not a bug (apologies, I couldn't work out how to add a question tag). I'm just wondering if this library, and indeed the IN219 device are suitable for use with a 4-20mA current loop, where the 4-20mA range represent 0 to 5A? If so, what would be the most appropriate settings for the library? Currently, my design is based on a 100 ohm shunt resistor, however, I could change it if there is a more appropriate resistor value.

Many thanks, and wishing you all the best for 2025!

Mark

RobTillaart, 2024-12-29:
@Mark-Wills 

If I understand correctly you have a "signal" wire from a sensor that provides 4-20 mA that represents 0.0 - 5.0 Amps

```
Signal    Current
4 mA      0.00 Amps
8 mA      1.25 Amps
12 mA     2.50 Amps
16 mA     3.75 Amps
20 mA     5.00 Amps
````

in formula  ```Current = 5.0 * (Signal - 4.0) / 16;```

I think measuring 4-20mA is a very low range for the INA sensor.  Hard to get precise.

If your processor has an ADC free, you could put a resistor in the signal wire.
- 150 Ω (1% or better)  => creates a voltage of 0.6 .. 3.0 Volt.  (3.3V processor) ```Current = 5 * voltage - (0.6) / (3.3 - 0.6);```
- 250 Ω (1% or better)  => creates a voltage of 1.0 .. 5.0 Volt.  (5.0V processor) ```Current = 5 * voltage - (1.0) / (5.0 - 1.0);```

Would that not be a simpler solution?

RobTillaart, 2025-01-04:
@Mark-Wills 

All the good for 2025, "may your dreams get wings" (African proverb).

Any progress / insights to report?

Mark-Wills, 2025-01-04:
Hi Rob

Many thanks for your swift response. In the end I went with a 16-bit ADC
and 100R burden resistor which should do the job just fine. This issue can
be closed.

Many thanks for looking at this, and my best wishes to you for 2025!

Mark

RobTillaart, 2025-01-04:
You're welcome!

Best wishes to you to, "May your dreams get wings!", (African proverb)

closed 2025-01-04 by RobTillaart