RobTillaart/INA219 #25: Reading low current values

arifainchtein, 2025-03-21:
Hello Rob,
I am using a custom made esp32 board with an ina219 and shunt resistor of .05ohms to measure the current coming out of 2 1.5F 5.5V capacitors. The board is powered by an usb connector connected to my computer.  I am including an image of the relevant part of the design.  I was trying the INA_DEMO example but i am getting very low values:

17:33:27.914 -> 	BUS		SHUNT		CURRENT		POWER		OVF		CNVR
17:33:27.914 -> 	4.44		0.11		             2.20		          9.16		0		1
17:33:28.907 -> 	4.44		0.10		          2.01		9.16		0		1
17:33:29.900 -> 	4.44		0.12		           2.41		10.99		0		1
17:33:30.925 -> 	4.44		0.01		           2.20		9.77		0		0
17:33:31.918 -> 	4.44		0.03		           1.80		7.93		0		0
17:33:32.911 -> 	4.44		0.01		          1.59		6.71		0		0
17:33:33.937 -> 	4.44		0.04		         1.59		6.71		0		0
17:33:34.930 -> 	4.44		0.08		         1.59		6.71		0		0

I cant believe that an esp32 is only consuming 2.20mA.  Is that because the esp32 is being fed by the usb port and there is not really that much current coming out of the capacitors?
in the Demo example, i set the following line:

  INA.setMaxCurrentShunt(1, 0.05);

because i am saying maximum is 1amp and the shunt resistor is 0.05

the line  Serial.println(INA.getBusVoltageRange()); returns 32.  isnt that a bit high given that the maximum voltage is going to be 5.5v?

Any suggestions will be greatly appreciated
thanks

[image]

RobTillaart, 2025-03-21:
Thanks for the question,

> Any suggestions will be greatly appreciated

higher quality images?  

Lets look at the output:

```cpp
    Serial.print("\t");
    Serial.print(INA.getBusVoltage(), 2);
    Serial.print("\t\t");
    Serial.print(INA.getShuntVoltage_mV(), 2);
    Serial.print("\t\t");
    Serial.print(INA.getCurrent_mA(), 2);
    Serial.print("\t\t");
    Serial.print(INA.getPower_mW(), 2);
    Serial.print("\t\t");
    Serial.print(INA.getMathOverflowFlag());
    Serial.print("\t\t");
    Serial.print(INA.getConversionFlag());
    Serial.println();
```

The shunt is 0.05 Ohm and it sees 0.11 milliVolt.
0.11 mV / 0.05 Ohm ==> 2.2 mA so that is consistent with what the library states. 

----

> I cant believe that an esp32 is only consuming 2.20mA. Is that because the esp32 is being fed by the usb port and there is not really that much current coming out of the capacitors?

You might tell that if you disconnect the USB port.
Use e.g. a 2nd processor or an RS232-TTL 2USB dongle to read the output of the sketch .

I assume that measuring the current from the capacitors is difficult, so you might try to  measure the current over USB with a multimeter. If you have a spare USB, cut it carefully in half, strip the wires and find the Vcc or GND wire to measure current.

There also exist USB power dongles that can measure the current e.g. https://www.tinytronics.nl/en/tools-and-mounting/measuring/measuring-modules/keweisi-usb-current-and-voltage-analyzer-with-lcd

Finally:
The ESP32 might not be that busy, found this interesting link that shows that 2.2 mA is not unusual low depending on which internal modules are used - https://lastminuteengineers.com/esp32-sleep-modes-power-consumption/

----

the line Serial.println(INA.getBusVoltageRange()); returns 32. isnt that a bit high given that the maximum voltage is going to be 5.5v?

No, read the datasheet of the INA219 (datasheets answers many questions)

[image]

arifainchtein, 2025-03-23:
Hey Rob,
Thanks for the answer, my apologies for the unreadable schematic.  I am working on some code to do store the data in ram sp and then extract it so that i can see a more detailed curve during solar panel.  thanks again, will keep you posted

RobTillaart, 2025-03-26:
@arifainchtein 

If the problem is solved, you may close the issue.

You can still add comments to share, or reopen if needed

RobTillaart, 2025-04-01:
As there is no answer I close the issue.
Feel free to reopen if needed

closed 2025-04-01 by RobTillaart